zaxour/TurboNanza #1: Core Grafx DIN mounting board minor issues

bazookaben, 2023-06-24:
I have a Core Grafx and noticed that in Step 5 you didn't mention to scrape away the solder mask on the two sides like for the Super Grafx. The layout of solder points in the AV connector area is identical to Super Grafx so I assume you just forgot to mention that in the Core Grafx section.

Also, there's a capacitor very close by that prevents me from perfectly centering the mounting board. Only off by a few millimeters probably, so no big deal. But thought I'd let you know about that too in case you want to make a slightly smaller mounting for Core Grafx in the future

zaxour, 2023-06-25:
Hi Ben,

Thanks for contacting me. I have updated the readme since I finally have a
CoreGrafx of my own to take pictures with. That large 470uF capacitor next
to the DIN is made unnecessary by the TurboNanza, so it can be removed
safely. Please take a look:

https://github.com/zaxour/TurboNanza#coregrafx

Thanks,
Zack

zaxour, 2023-06-26:
> Thank you
>
> I had another question about the bottom shell. You mentioned that it had
> two ribs that needed to be trimmed. Maybe you’re thinking about the
> original PC Engine?
>
> I didn’t see those ribs but I had four plastic cylinders that were holding
> spring loaded round tabs for the expansion port. So I cut those off and
> removed the springs and metal. I was hoping to leave the case more stock
> condition than that but oh well
>
> But was that the right way to go about that? I couldn’t think of a better
> way to get the two halves of shell to close

Hi Ben,

You're absolutely correct. This is something else I will be adding more
information on as I do the mod on the CG.

Thanks,
Zack

zaxour, 2023-06-26:
> Cool deal
>
> And have you had a chance to look at the audio presets for the Turbo
> Everdrive Pro and how they sound with the Turbonanaza? Whether there needs
> to be custom volume levels set to get a close match to NEC’s CD systems?
>
> I see that there is a “Core AV” and “Core EXP” presets. I imagine the Core
> EXP is closer to what we need to but I imagine it’s tailored specifically
> to Krikkz’s EDFX add on. Like it has left channel ay 100% volume and right
> channel at 88%, if I recall correctly

I use the direct option, which sets the L/R audio channels to 100%, I
believe. I don't touch the CDDA/PCM values. I haven't done an MDFourier
analysis on it yet to figure out what the best presets are.

Thanks,
Zack