zaxour/TurboNanza #5: SSDS3 Incompatibility

Peredonov, 2023-12-26:
I tried to hook up an old Terraonion SSDS3 to a Turbonanza'd PC Engine to load CD games from the former while using S-video from the latter, but the SSDS3 cannot start (either displays a white screen or a green screen with garbled graphics and audio noise). The SSDS3 works perfectly on non-Turbonanza consoles (and it has always worked with consoles modded with other RGB mods). The Turbonanza'd console works well otherwise (and is compatible with a Super CD-ROM attachment).

It would be good for someone with both an SSDS3 and a Turbonanza mod to confirm.

zaxour, 2023-12-26:
That’s really strange. I have personally tested an SSDS3 on a bonanza system and it has worked fine. The bonanza doesn’t touch any of the digital bus lines, so I’m not sure what would be causing the issue. I’ll try the ssds with a few systems and see what the results are.

Thanks,
Zack

Peredonov, 2023-12-27:
Thanks for checking. The console that has this problem for me is a white PCE, the SSDS3 powers on, green LED turns on, but it seems like the menu system is not able to load. 

Please let me know if you have any ideas about anything on the PCE board that could have gotten messed up during the bonanza install that might be interfering with the SSDS3 loading.

zaxour, 2024-01-21:
Sorry for the delay. I just double checked SSDS3 compatibility on my CoreGrafx and it was working fine. You may want to make sure that the SSDS3 is fully seated. Further, you can use the bonanza video output when using SSDS3 to take advantage of svideo.

Zack

projectzero, 2024-05-14:
I had the same problem with my recent install on a CoreGrafx - white/green garbled screen and audio noise. Interestingly, the graphics were fine if I used an Everdrive 2.5 and bypassed the flashcard capabilities of the SSDS3 (but still used it's video out). I fixed the issue by removing the bottom shielding around the expansion port pins (did keep the 2 pins that connect near the port) - previously I had left this shielding off as I'm using one of the aftermarket clear shells. 

I do have a noticeable audio hum/buzz when using the SSDS3 flashcart capabilities and video out through the Turbonanza (SCART or HDRetrovision cables). I don't *think* it's an SD card issue as the hum is constant and not just during disk access but since I solved the bigger issue (SSDS3 flashcard), I'll troubleshoot the hum later.