zaxour/TurboNanza #7: Can't SVIDEO and RGB coexist?

Cyrus331, 2024-01-12:
My PCEINGINE model is TG001. Can't SVIDEO and RGB coexist? Step 4 - Solder S-Video connections. After I operated, SVIDEO could be used, but there was no RGB signal output. Later, I did not remove those resistors and directly connected the SVIDEO terminals to the corresponding solder joints. The result can coexist, that is, both It can output SVIDEO or RGB, but today when I used SVIDEO to output, the picture suddenly became dark, as if there was a lack of brightness, but the colors were still there. I wonder if it was because I did not remove the resistor?

When I tried removing any of the resistors R145, R141 or R134, there was no RGB signal output, so I had to keep these resistors.
The dark picture is SVIDEO, and the bright picture is normal AV picture.
[image]
[image]

[image]

Cyrus331, 2024-01-12:
I followed the steps again to remove those resistors, but the SVIDEO picture was still dark.
[image]

zaxour, 2024-01-13:
Hi,

S-video and RGB can absolutely coexist. Did you build the bonanza yourself?
If so, I would go back and make sure that proper component values were
used. What cables are you using for RGB? The sync signal from the PCE is
fairly low, so some cables might not have enough margin - I would suggest
Sync on Luma if possible.

Also, the FFC is very difficult to hand solder. I would recommend reflowing
if possible.

Zack

Cyrus331, 2024-01-14:
Thanks for the reply. Now the problem is solved when I connect the Y signal to R145. If the same effect can be achieved without removing the resistors, why were those resistors disconnected in the original design? It's because The sync signal from the PCE is
Fairly low, so some cables might not have enough margin?

zaxour, 2024-01-14:
I found that the svideo signal was too noisy if I tapped the signals directly, so I reimplemented the whole circuit on my board. How are you connecting your cable? If everything is going through the bonanza output, I don’t see how your TV would be pulling the signals low. The only time I’ve seen that is if using one of those adapters from AliExpress where composite video is not buffered. This causes the composite signal to be loaded down. It doesn’t happen when using a proper output device.

Zack