zaxour/TurboNanza #9: New TurboNanza install black screen and buzzing sound

TruREALigion, 2024-05-03:
I installed on PC Engine but no picture and loud buzzing sound.  Wondering what troubleshooting steps I can try?  

It works fine when I plug into my IFU unit.

[image]
[image]
[image]
[image]

zaxour, 2024-05-03:
Hi,

Can you show me how you installed the DIN? I've seen people install it
upside down, just want to make sure that's not the case.

Thanks!

TruREALigion, 2024-05-03:
I didn’t install the DIN actually.  It is just hanging outside.  Does it need to be mounted in for it to work?

zaxour, 2024-05-03:
It doesn’t need to be mounted to the system to work. However, I would like to confirm how you’ve soldered on the small DIN board.

TruREALigion, 2024-05-03:
See attached

[image]

[image]

zaxour, 2024-05-03:
> See attached
> image.jpg (view on web)
> IMG_8880.jpeg (view on web)

The din board is upside down, that would be why you’re not getting an output. Please desolder and flip it over.

TruREALigion, 2024-05-03:
That explains everything!

Will update back and close if resolved.

TruREALigion, 2024-05-03:
Fixed.  I had the DIN upside-down.

closed 2024-05-03 by TruREALigion